thotro/arduino-dw1000 #343: Adding more tags to the system

anthony211212, 2024-04-28:
Hello. Has anyone fully solved the issue of adding multiple tags? I've looked into discussions among the question titles here, but I couldn't find a precise and clear solution. Has anyone been able to use this library for multiple anchors and multiple tags for example 4 anchors and 4 tags or more? Or perhaps, does anyone know of a different library suitable for this situation? Your recommendations are highly valuable.

az-z, 2024-04-28:
Anthony,
 My last experience with this lib was 7ish years ago. I couldn't get it to
work then. That is why I moved to some other lib and esp32 type modules.
I have not been in iot scene for good 3 years now.
I'm sorry I can't be of more help.

anthony211212, 2024-04-28:
Hello. I know some modules ESP32 UWB(Ultra Wideband) and ESP32 UWB DW3000(Ultra Wideband). Do you know any library for them to use multiple tags and anchors ?

scal68, 2024-05-30:
Bonjour,
Je travail actuellement sur la même problématique. Avez-vous trouvez une solution?
Ou même une façon de limiter le nombres de balises qui ce connectes de façon "propre"?

anthony211212, 2024-07-26:
Hello. Not yet. Do you have any idea?

pizzo00, 2024-09-11:
**Update for anyone who encounter this problem**
You can find an updated version that fixes this and other problems here: https://github.com/jremington/UWB-Indoor-Localization_Arduino/tree/main/DW1000_library_pizzo00

anthony211212, 2024-09-11:
@pizzo00 Which algorithm did you use to find the tag's position, and what was your error value?

pizzo00, 2024-09-11:
The trilateration algorithm is beyond the scope of the library but I personally use the @jremington [implementation](https://github.com/jremington/UWB-Indoor-Localization_Arduino/blob/main/ESP32_UWB_tag3D_4A/ESP32_UWB_tag3D_4A.ino) of this [paper](https://www.th-luebeck.de/fileadmin/media_cosa/Dateien/Veroeffentlichungen/Sammlung/TR-2-2015-least-sqaures-with-ToA.pdf) with a RMSE of 2m on a distance of 30m from the anchors

scal68, 2024-09-15:
@pizzo00 Merci beaucoup pour votre travail. Cependant, je ne parviens pas a utiliser votre bibliothèque, La compilation est OK avec le Biblio d'origine, mais j'ai de nombreuse erreur avec la votre. Devons-nous modifier des éléments dans notre script?

pizzo00, 2024-09-16:
@scal68 Now it is sufficient to call the function DW1000Ranging.init, with the type, the short address, a mac address, a flag for high_power, the transmission mode, and the usual parameters.

My code looks like this:

```
static constexpr byte MY_MODE[] = {DW1000.TRX_RATE_6800KBPS, DW1000.TX_PULSE_FREQ_16MHZ, DW1000.TX_PREAMBLE_LEN_64};

void setup()
{
    Serial.begin(115200);
    delay(1000);

    uint16_t shortMacAddress = 118;  //"76:00";
    char *wifiMacAddress = "00:00:00:AA:AA:AA"; //I get it from the esp board through WiFi.macAddress();

    // init the configuration
    SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
#if C_TAG
    DW1000Ranging.init(BoardType::TAG, shortMacAddress, wifiMacAddress, true, MY_MODE, PIN_RST, PIN_SS, PIN_IRQ);
#elif C_ANCHOR
    DW1000Ranging.init(BoardType::ANCHOR, shortMacAddress, wifiMacAddress, true, MY_MODE, PIN_RST, PIN_SS, PIN_IRQ);
#endif
}
```

johnreb20, 2025-06-23:
@pizzo00 so what have to change in the ESP32_UWB_tag2D_3A of @jremington implementation?